SteveMacenski/slam_toolbox #735: No map in RVIZ livox mid360

maxxbolotov, 2024-12-02:
Hi everyone, I would be very grateful for the help.
Ubuntu 22.04, ros2 humble

I'm using Livox mid360 for generating PointClouds and pointclouds_to_laserscan to get a laserscan. 
So then I'd want to use online_async_launch.py to create a 2d map. But I after I execute it I have no map under the topic /map in RVIZ.
I guess I have to associate my laserscan topic with online_async_launch but I don't know where and what to change.

Thank you.

maxxbolotov, 2024-12-02:
upd:
After i set my laserscan topic name as default /scan i start to receive such messages in the terminal i used to launch online_async_launch.py:
async_slam_toolbox_node-1] [INFO] [1733162428.557220152] [slam_toolbox]: Message Filter dropping message: frame 'body' at time 1733162427.153 for reason 'discarding message because the queue is full

SteveMacenski, 2024-12-02:
This is better suited for Robotics Stack Exchange & possibly googling to see if others have run into similar issues as you for you to find where your issue lies. Please read the SLAM Toolbox readme on the required input topics / TF transformations and generally make sure that your ROS network is correctly configured (i.e. driver is publishing, laser scan is properly formatted, etc)

closed 2024-12-02 by SteveMacenski