SteveMacenski/slam_toolbox #736: srvs not accessible in Apt install on Humble

vinnnyr, 2024-12-05:
**Required Info:**

- Operating System:
Ubuntu 22.04,
- Installation type:
  Apt Install 
- ROS Version
ROS Humble
- Version or commit hash:
```
Package: ros-humble-slam-toolbox
Version: 2.6.8-1jammy.20240830.234419
```
- Laser unit:
N/A
  
#### Steps to reproduce issue
`ros2 interface show slam_toolbox/srv/SaveMap`  OR `ros2 service call slam_toolbox/srv/SaveMap /slam_toolbox/save_map` 

#### Expected behavior
We get the service definition and are able to use the service.

#### Actual behavior
for `ros2 interface show slam_toolbox/srv/SaveMap` 
```
Error processing '// generated from rosidl_adapter/resource/srv.idl.em' of 'slam_toolbox/SaveMap': '//'
Traceback (most recent call last):
  File "/opt/ros/humble/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.18.11', 'console_scripts', 'ros2')())
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2cli/cli.py", line 91, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2interface/command/interface.py", line 35, in main
    return extension.main(args=args)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2interface/verb/show.py", line 201, in main
    _show_interface(
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2interface/verb/show.py", line 148, in _show_interface
    for line in _get_interface_lines(interface_identifier):
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2interface/verb/show.py", line 116, in _get_interface_lines
    yield InterfaceTextLine(
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2interface/verb/show.py", line 43, in __init__
    msg_spec = parse_message_string(
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/parser.py", line 520, in parse_message_string
    Type(type_string, context_package_name=pkg_name),
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/parser.py", line 277, in __init__
    super(Type, self).__init__(
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/parser.py", line 190, in __init__
    raise InvalidResourceName(type_string)
rosidl_adapter.parser.InvalidResourceName: //
```

for `ros2 service call slam_toolbox/srv/SaveMap /slam_toolbox/save_map "{name: '/home/foobar/Downloads/test_map'}"` 
```
Traceback (most recent call last):
  File "/opt/ros/humble/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.18.11', 'console_scripts', 'ros2')())
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2cli/cli.py", line 91, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2service/command/service.py", line 41, in main
    return extension.main(args=args)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2service/verb/call.py", line 58, in main
    return requester(
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2service/verb/call.py", line 69, in requester
    module = importlib.import_module('.'.join(parts[:-1]))
  File "/usr/lib/python3.10/importlib/__init__.py", line 121, in import_module
    raise TypeError(msg.format(name))
TypeError: the 'package' argument is required to perform a relative import for '.slam_toolbox'
```

#### Additional information
This was reproduced across multiple machines with similar environments.  I haven't yet been able to see if a source install fixes this issue.

vinnnyr, 2024-12-05:
I was able to reproduce on a fresh ros:humble container as well

SteveMacenski, 2024-12-05:
Are you sure its a problem with slam toolbox, or perhaps its a regression in the ros cli, idl tools? Do you have a problem compiling or calling the service from another piece of software?

vinnnyr, 2024-12-05:
I was able to confirm this happens in a source install as well inside a fresh ros:humble container. 

I spot checked ros2 interface show for other installled services and they work fine
```
root@4f192e78d452:~/slam_ws/src/slam_toolbox# ros2 interface show std_srvs/srv/Trigger 
---
bool success   # indicate successful run of triggered service
string message # informational, e.g. for error messages

```

so not sure if it's a regression in ros2 cli. If it is, it isn't a large system-wide regression

SteveMacenski, 2024-12-05:
Yeah I see the same behavior. Can you call the service from some C++/python code though? It would help narrow it down if we knew if it was just isolated to the CLI or was a global issue that **none** of the services can be caleld

SteveMacenski, 2024-12-05:
Looking at robot localization as an example that seems to work (since its another rare example of interfaces within another C++ package) and I see the following things that aren't in slam toolbox, maybe one of these creates an issue?
- https://github.com/cra-ros-pkg/robot_localization/blob/ros2/package.xml#L43
- https://github.com/cra-ros-pkg/robot_localization/blob/ros2/package.xml#L18
- https://github.com/cra-ros-pkg/robot_localization/blob/ros2/CMakeLists.txt#L363C3-L363C25
- I also see multiple `rosidl_get_typesupport_target` calls, but so does R_L

vinnnyr, 2024-12-05:
Oh interesting, this minimal python service call works
```
#!/usr/bin/env python3

import rclpy
from rclpy.node import Node
from slam_toolbox.srv import SaveMap

class MapSaver(Node):
    def __init__(self):
        super().__init__("map_saver")
        self.client = self.create_client(SaveMap, "/slam_toolbox/save_map")
        while not self.client.wait_for_service(timeout_sec=1.0):
            self.get_logger().info("Service not available, waiting...")

        self.save_map()

    def save_map(self):
        request = SaveMap.Request()
        request.name.data = "/home/vinny/foobar_map"

        future = self.client.call_async(request)
        rclpy.spin_until_future_complete(self, future)

        if future.result() is not None:
            self.get_logger().info("Map saved successfully")
        else:
            self.get_logger().error("Failed to save map")

def main():
    rclpy.init()
    map_saver = MapSaver()
    rclpy.spin(map_saver)
    map_saver.destroy_node()
    rclpy.shutdown()

if __name__ == "__main__":
    main()

```

SteveMacenski, 2024-12-05:
That makes sense to me - else someone would have reported / complained about this long ago. I'm guessing something just isn't exported correctly that the CLI relies on

vinnnyr, 2024-12-05:
Interesting. Given that information, I am happy to drop the issue, I just wasn't expecting the CLI to not work but have the minimal program work.

vinnnyr, 2024-12-05:
But as an aside, neither of those diffs solved the issue
```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index aab4ab8..9b4ba53 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -41,6 +41,7 @@ set(CMAKE_INCLUDE_CURRENT_DIR ON)
 
 set(dependencies
   rclcpp
+  rosidl_default_runtime
   message_filters
   nav_msgs
   sensor_msgs
diff --git a/package.xml b/package.xml
index 17dee6f..d5cb750 100644
--- a/package.xml
+++ b/package.xml
@@ -12,6 +12,7 @@
   <author>Steve Macenski</author>
 
   <buildtool_depend>ament_cmake</buildtool_depend>
+  <buildtool_depend>rosidl_default_generators</buildtool_depend>
 
   <build_depend>pluginlib</build_depend>
   <build_depend>eigen</build_depend>
@@ -65,6 +66,7 @@
   <exec_depend>nav2_map_server</exec_depend>
   <exec_depend>builtin_interfaces</exec_depend>
   <exec_depend>rosidl_default_generators</exec_depend>
+  <exec_depend>rosidl_default_runtime</exec_depend>
 
   <exec_depend>libqt5-core</exec_depend>
   <exec_depend>libqt5-gui</exec_depend>

```

SteveMacenski, 2024-12-05:
Odd - ok. Given the software works, I'm going to mark this as low-priority, but going to leave the ticket open since it is a bug. I may get to it over the holidays as I'm looking for something to do on the plane, but no promises.

If another person reads this and wants to take it on, please! I imagine its something small.

kei1107, 2024-12-10:
It was a small thing. :)
The `.srv` file is placed in the `srvs` directory, not the `srv` directory.

<https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Custom-ROS2-Interfaces.html#create-a-new-package>
> The .msg and .srv files are required to be placed in directories called msg and srv respectively.

```console
$ ros2 interface show slam_toolbox/srv/SaveMap 
std_msgs/String name
        string data
---
# Result code defintions
uint8 RESULT_SUCCESS=0
uint8 RESULT_NO_MAP_RECEIEVD=1
uint8 RESULT_UNDEFINED_FAILURE=255

uint8 result
```

kei1107, 2024-12-10:
@vinnnyr 

> ros2 service call slam_toolbox/srv/SaveMap /slam_toolbox/save_map "{name: '/home/foobar/Downloads/test_map'}"

It looks like the service_name and service_type are reversed.

`ros2 service call /slam_toolbox/save_map slam_toolbox/srv/SaveMap "name: {data: '/home/foobar/Downloads/test_map'}"`

```console
$ ros2 service call /slam_toolbox/save_map slam_toolbox/srv/SaveMap "name: {data: '/home/foobar/Downloads/test_map'}"
requester: making request: slam_toolbox.srv.SaveMap_Request(name=std_msgs.msg.String(data='/home/foobar/Downloads/test_map'))

response:
slam_toolbox.srv.SaveMap_Response(result=1)
```

SteveMacenski, 2024-12-10:
Niiiice - looks good to me!

SteveMacenski, 2024-12-11:
Merged -- I'll be running releases on Friday and will make sure to do slam toolbox. Thanks for the report @vinnnyr and the debugging @kei1107 !

closed 2024-12-11 by SteveMacenski

vinnnyr, 2024-12-11:
> It looks like the service_name and service_type are reversed.

🙄  definitely a case of me turning off my brain after one related thing was not working haha

Good find! At least the original issue was somewhat real :)