SteveMacenski/slam_toolbox #742: SlamToolboxPlugin ros2 humble can't save map files .data, .posegraph, .pgm, .yaml

wladimirpetrov, 2025-01-09:
**Required Info:**

- Operating System:
  - Ubuntu 22.04.5 LTS
- Installation type:
  - ROS 2 packages and CrazySwarm2 installed from source
- ROS Version
  - ROS 2 Humble
- Version or commit hash:
  - Using the latest release of Slam Toolbox and CrazySwarm2 from their respective repositories.
  
#### Steps to reproduce issue
1. Start the Crazyflie server using the configured crazyflies.yaml file and the multiranger_nav2_launch.py launch file.
2. Open RViz2 and add the SLAM Toolbox plugin.
3. Attempt to save the map using the "Save Map" button in the SLAM Toolbox plugin interface.
4. Alternatively, try saving the map via the CLI:
`ros2 service call /slam_toolbox/save_map slam_toolbox/srv/SaveMap "{name: '~/ros2_ws/maps/map'}"
`
5. Another attempt was made using:
`ros2 run nav2_map_server map_saver_cli -f ~/ros2_ws/maps/map --ros-args -r map:=/cf231/map -p map_subscribe_transient_local:=true
`

#### Expected behavior
When saving the map using either the SLAM Toolbox plugin in RViz2 or via the CLI, the following files should be generated:
.pgm
.yaml
.data
.posegraph
These files are required for localization and further map-related tasks.

#### Actual behavior
1. Clicking the "Save Map" button in the SLAM Toolbox plugin does nothing. The terminal shows the following warning:
`[WARN] [SlamToolboxPlugin]: SlamToolbox: Failed to save map as , is service running?
`
2. Attempting to call the /slam_toolbox/save_map service results in the following error:
`Failed to populate field: Value '~/ros2_ws/maps/map' is expected to be a dictionary but is a str
`
3. Using the nav2_map_server map_saver_cli command successfully generates the .pgm and .yaml files, but the .data and .posegraph files are not generated.

#### Additional information
Here is the list of nodes running during the process:
`/SlamToolboxPlugin
/SlamToolboxStateUpdateNode
/camera/camera
/crazyflie_server
/rviz
/teleop_twist_keyboard
/transform_listener_impl_613584c3b3b0
/vel_mux
`
List of active topics:
`/cf231/cmd_full_state
/cf231/cmd_hover
/cf231/cmd_vel_legacy
/cf231/map
/cf231/map_updates
/cf231/odom
/cf231/pose
/cf231/robot_description
/cf231/scan
/cf231/status
/cmd_vel
/initialpose
/parameter_events
/poses
/rosout
/tf
/tf_static
`
Conclusion
While the SLAM Toolbox is able to generate .pgm and .yaml files using the nav2_map_server map_saver_cli, the .data and .posegraph files are not being created. Using the SLAM Toolbox plugin in RViz2 also fails to save the map entirely, with the warning message indicating the service might not be running

SteveMacenski, 2025-01-09:
> name: '~/ros2_ws/maps/map'

Did you try to use full filepaths? I think it doesn't understand `~/`. 

PS the slam toolbox service just calls the map server, so 4/5 are identical operations that will have the same result.

wladimirpetrov, 2025-01-10:
> > name: '~/ros2_ws/maps/map'
> 
> Did you try to use full filepaths? I think it doesn't understand `~/`.
> 
> PS the slam toolbox service just calls the map server, so 4/5 are identical operations that will have the same result.

I tried with full filepath: 

`ros2 service call /slam_toolbox/save_map slam_toolbox/srv/SaveMap "{name: '/home/petrov/ros2_ws/maps/map'}"
Failed to populate field: Value '/home/petrov/ros2_ws/maps/map' is expected to be a dictionary but is a str
`
I thought there might be an issue with permission saving/writing, but it turns out it has all permissions

SteveMacenski, 2025-01-10:
My autocomplete looks like: 

```
ros2 service call /hi slam_toolbox/srv/SaveMap "name:
  data: ''"
```

Are you sure that this is the correct format? It seems like a CLI formatting issue - that error is not coming from SLAM Toolbox, but your terminal sending the CLI command. I think the command is invalid by the message type

wladimirpetrov, 2025-01-10:
oh... yes, you're absolutely right, thank you!
Now when I run:
`ros2 service call /slam_toolbox/save_map slam_toolbox/srv/SaveMap "name:
  data: '/home/petrov/ros2_ws/maps/map'"
`
in other terminal i get: 
`ros2 launch slam_toolbox online_async_launch.py
[INFO] [launch]: All log files can be found below /homes/petrov/.ros/log/2025-01-10-15-18-03-170354-intern97dt-142704
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [async_slam_toolbox_node-1]: process started with pid [142705]
[async_slam_toolbox_node-1] [INFO] [1736540283.203505848] [slam_toolbox]: Node using stack size 40000000
[async_slam_toolbox_node-1] [INFO] [1736540283.216056221] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[async_slam_toolbox_node-1] [INFO] [1736540283.216121805] [slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[async_slam_toolbox_node-1] [WARN] [1736540305.030248774] [slam_toolbox]: Map Saver: Cannot save map, no map yet received on topic /map.
`
different error but still doesn't want to save the map

wladimirpetrov, 2025-01-13:
nothing is being published to /map topic for some reasons

SteveMacenski, 2025-01-13:
Well, that would be your issue :-)

wladimirpetrov, 2025-01-14:
Well, could you help me our what the reason could be then for that?
I mean now playing with 1 crazyflie and the ros2 topic list is:
`ros2 topic list
/cf231/cmd_full_state
/cf231/cmd_hover
/cf231/cmd_vel_legacy
/cf231/map
/cf231/map_updates
/cf231/odom
/cf231/pose
/cf231/robot_description
/cf231/scan
/cf231/status
/clicked_point
/clock
/cmd_vel
/goal_pose
/initialpose
/map
/map_metadata
/parameter_events
/pose
/poses
/rosout
/scan
/slam_toolbox/feedback
/slam_toolbox/graph_visualization
/slam_toolbox/scan_visualization
/slam_toolbox/update
/tf
/tf_static`
all the topics that start with cf231 publish correctly, however when it comes to SLAM topics (/map etc all of them don't publish anything).

wladimirpetrov, 2025-01-14:
the problem was solved by several re-installations of slam toolbox for ROS2 humble. 
Sorry for bothering you. Thank you for your help!

closed 2025-01-14 by SteveMacenski