SteveMacenski/slam_toolbox #744: Using slam_toolbox for localization

Agri-Mechatronic, 2025-01-14:
- Operating System:
  - Ubuntu 22.04
- Installation type:
  - 
- ROS Version
  - ROS2 Humble

Hello,

I am using turtlebot3 simulation to simulate localization in pre-saved and serialized map. I am experiencing issues with slam_toolbox in localization mode while using it in a simulation environment. Initially, the localization works correctly, but after a few seconds, the robot's pose becomes inconsistent, and it seems to jump to random position on the map. The following points describe the problem in detail:

I have ensured that there are no **inf** values in the laser scan ranges.
However, I still see the following warning repeatedly in the logs:

Info: clipped range threshold to be within minimum and maximum range!
I wonder if this could be affecting the scan matching process and leading to incorrect localization.

**TF Tree and Topic Rates:**

The TF tree looks correct, and the parent-child relationships are properly defined.
Topic rates are as follows:
/filtered_scan: ~10 Hz
/odom: ~29 Hz
/tf: ~49 Hz
Despite these seemingly correct rates, the localization eventually fails.

**Attempted Solutions:**

I have followed the discussions in related GitHub issues (#720, #611, #576, #465, #422, #415) but have not found a solution that resolves my problem.
I tried using debug logs and tools like swri_console to understand the issue better, but nothing conclusive was found.

Launch Command:

ros2 launch slam_toolbox localization_launch.py params_file:=/path/to/mapper_params_localization.yaml

mapper_params_localization.yaml:

```
slam_toolbox:
  ros__parameters:
    solver_plugin: solver_plugins::CeresSolver
    ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
    ceres_preconditioner: SCHUR_JACOBI
    ceres_trust_strategy: LEVENBERG_MARQUARDT
    ceres_dogleg_type: TRADITIONAL_DOGLEG
    ceres_loss_function: None

    # ROS Parameters
    use_sim_time: true
    mode: "localization"
    map_file_name: "/home/user/ros2_ws/maps/map_for_localization"
    scan_topic: /filtered_scan
    base_frame: base_footprint
    odom_frame: odom
    map_frame: map
    transform_publish_period: 0.1
    map_update_interval: 0.5
    transform_tolerance: 0.1

    # if you'd like to start localizing on bringup in a map and pose
    #map_file_name: test_steve
    map_start_pose: [0.0, 0.0, 0.0]

    debug_logging: false
    throttle_scans: 1
    transform_publish_period: 0.05 #if 0 never publishes odometry
    map_update_interval: 0.0
    resolution: 0.05
    min_laser_range: 0.15 #for rastering images
    max_laser_range: 3.5 #for rastering images
    minimum_time_interval: 0.5
    transform_timeout: 0.2
    tf_buffer_duration: 100.0
    stack_size_to_use: 40000000 #// program needs a larger stack size to serialize large maps

    # General Parameters
    use_scan_matching: true
    use_scan_barycenter: true
    minimum_travel_distance: 0.2
    minimum_travel_heading: 0.2
    scan_buffer_size: 5
    scan_buffer_maximum_scan_distance: 8.0
    link_match_minimum_response_fine: 0.1  
    link_scan_maximum_distance: 1.5
    do_loop_closing: true 
    loop_match_minimum_chain_size: 3
    loop_match_maximum_variance_coarse: 3.0  
    loop_match_minimum_response_coarse: 0.35    
    loop_match_minimum_response_fine: 0.45
    
    scan_queue_size: 50
    scan_qos_history: "keep_last"
    scan_qos_reliability: "reliable"
    scan_qos_durability: "volatile"
    num_threads: 4

    # Correlation Parameters - Correlation Parameters
    correlation_search_space_dimension: 0.5
    correlation_search_space_resolution: 0.01
    correlation_search_space_smear_deviation: 0.1 

    # Correlation Parameters - Loop Closure Parameters
    loop_search_space_dimension: 8.0
    loop_search_space_resolution: 0.05
    loop_search_space_smear_deviation: 0.03
    loop_search_maximum_distance: 3.0

    # Scan Matcher Parameters
    distance_variance_penalty: 0.5      
    angle_variance_penalty: 1.0    

    fine_search_angle_offset: 0.00349     
    coarse_search_angle_offset: 0.349   
    coarse_angle_resolution: 0.0349        
    minimum_angle_penalty: 0.9
    minimum_distance_penalty: 0.5
    use_response_expansion: true
```

Thanks everyone...

SteveMacenski, 2025-01-14:
This is Nav2, not slam toolbox, please file with the correct repository.

Agri-Mechatronic, 2025-01-14:
Sorry but I could not understand, isn't this slam_toolbox issues page?

SteveMacenski, 2025-01-15:
Oh sorry, that's my bad misreading my notifications tray :laughing: 

> Info: clipped range threshold to be within minimum and maximum range!

It [occurs here](https://github.com/SteveMacenski/slam_toolbox/blob/45dfd229f8a4d245ee2f729f4437c459a85412ba/lib/karto_sdk/include/karto_sdk/Karto.h#L3952), which is called [from here](https://github.com/SteveMacenski/slam_toolbox/blob/45dfd229f8a4d245ee2f729f4437c459a85412ba/src/laser_utils.cpp#L171). What I can ascertain is that `max_laser_range` parameter (3.5) was clipped to be within the min/max ranges - so most likely your scan that's being published from the TB3 is > 3.5 range.

Agri-Mechatronic, 2025-01-19:
Thank you for your response;

Could this be the underlying cause of the localization issue I'm experiencing? When I start the localization process, the robot initially localizes itself correctly for a few seconds. However, shortly after, I see the message "[localization_slam_toolbox_node-1] Registering sensor: [Custom Described Lidar]" in the logs, and the robot's position on the map changes unexpectedly.

Below is the output with additional debug logs that I added for further insight:

```
[localization_slam_toolbox_node-1] Mapper <- Module
[localization_slam_toolbox_node-1] Mapper <- m_pSequentialScanMatcher
[localization_slam_toolbox_node-1] Mapper <- m_pGraph
[localization_slam_toolbox_node-1] MapperGraph <- Graph; Graph <- m_Edges; Graph <- m_Vertices
[localization_slam_toolbox_node-1] MapperGraph <- m_pMapper; MapperGraph <- m_pLoopScanMatcher; MapperGraph <- m_pTraversal
[localization_slam_toolbox_node-1] Mapper <- m_pMapperSensorManager
[localization_slam_toolbox_node-1] MapperSensorManager <- m_ScanManagers; MapperSensorManager <- m_Scans
[localization_slam_toolbox_node-1] Mapper <- m_Listeners
[localization_slam_toolbox_node-1] **Finished serializing Mapper**
[localization_slam_toolbox_node-1] Load From File
[localization_slam_toolbox_node-1] **Serializing Dataset**
[localization_slam_toolbox_node-1] Dataset <- m_SensorNameLookup
[localization_slam_toolbox_node-1] Dataset <- m_Data
[localization_slam_toolbox_node-1] Dataset <- m_Lasers
[localization_slam_toolbox_node-1] Dataset <- m_pDatasetInfo
[localization_slam_toolbox_node-1] **Finished serializing Dataset**
[localization_slam_toolbox_node-1] Registering sensor: [Custom Described Lidar]
[localization_slam_toolbox_node-1] WARNING: Logging before InitGoogleLogging() is written to STDERR
[localization_slam_toolbox_node-1] W0120 00:37:52.146977 60612 preprocessor.cc:62] Specified options.num_threads: 50 exceeds maximum available from the threading model Ceres was compiled with: 12.  Bounding to maximum number available.
[localization_slam_toolbox_node-1] [INFO] [1737329872.239306558] [slam_toolbox]: Laser scan received with 360 ranges
[localization_slam_toolbox_node-1] [INFO] [1737329872.239358746] [slam_toolbox]: First range value: 3.400000
[localization_slam_toolbox_node-1] [INFO] [1737329872.239370358] [slam_toolbox]: Constructor called in laserCallback
[localization_slam_toolbox_node-1] [WARN] [1737329872.239541118] [slam_toolbox]: minimum laser range setting (0.1 m) exceeds the capabilities of the used Lidar (0.1 m)
[localization_slam_toolbox_node-1] [INFO] [1737329872.239623543] [slam_toolbox]: Constructor called in addScan
[localization_slam_toolbox_node-1] Info: clipped range threshold to be within minimum and maximum range!
[localization_slam_toolbox_node-1] Registering sensor: [Custom Described Lidar]
[localization_slam_toolbox_node-1] [INFO] [1737329876.041543293] [slam_toolbox]: Laser scan received with 360 ranges
[localization_slam_toolbox_node-1] [INFO] [1737329876.041585242] [slam_toolbox]: First range value: 3.400000
[localization_slam_toolbox_node-1] [INFO] [1737329876.041595040] [slam_toolbox]: Constructor called in laserCallback
[localization_slam_toolbox_node-1] [INFO] [1737329876.042225084] [slam_toolbox]: Laser scan received with 360 ranges
[localization_slam_toolbox_node-1] [INFO] [1737329876.042249149] [slam_toolbox]: First range value: 3.400000
[localization_slam_toolbox_node-1] [INFO] [1737329876.042261462] [slam_toolbox]: Constructor called in laserCallback
[localization_slam_toolbox_node-1] [INFO] [1737329876.042496353] [slam_toolbox]: Laser scan received with 360 ranges
[localization_slam_toolbox_node-1] [INFO] [1737329876.042508787] [slam_toolbox]: First range value: 3.400000
```

I have created one node for limiting datas from lidar sensor and it outputs datas like:

```
header:
  stamp:
    sec: 19
    nanosec: 511000000
  frame_id: base_scan
angle_min: 0.0
angle_max: 6.28000020980835
angle_increment: 0.01749303564429283
time_increment: 0.0
scan_time: 0.0
range_min: 0.10000000149011612
range_max: 3.4000000953674316
ranges:
- 3.4000000953674316
- 3.4000000953674316
- 3.4000000953674316
- 3.4000000953674316
- 3.4000000953674316
- 3.4000000953674316
- 3.4000000953674316
- 3.1055243015289307
- 3.0426688194274902
- 3.0501654148101807
- 2.038742780685425
- 1.9977275133132935
- 1.985163688659668
```

And from the yaml file of localization, we can see min and max values:

```
    use_sim_time: true
    mode: "localization"
    map_file_name: "/home/user/ros2_ws/maps/map_for_localization"
    scan_topic: /filtered_scan
    base_frame: base_footprint
    odom_frame: odom
    map_frame: map
    transform_publish_period: 0.1
    map_update_interval: 0.5
    transform_tolerance: 0.1

    # if you'd like to start localizing on bringup in a map and pose
    #map_file_name: test_steve
    map_start_pose: [0.0, 0.0, 0.0]

    debug_logging: false
    throttle_scans: 1
    transform_publish_period: 0.05 #if 0 never publishes odometry
    map_update_interval: 0.0
    resolution: 0.05
    min_laser_range: 0.1 #for rastering images
    max_laser_range: 3.5 #for rastering images
    minimum_time_interval: 0.5
    transform_timeout: 0.2
    tf_buffer_duration: 100.0
    stack_size_to_use: 40000000 #// program needs a larger stack size to serialize large maps
```

But I can't still solve the problem with localization. I want turtlebot3 to localize itself in the pre-saved map without needing to set initial position from rviz or something. Weird part here is that localization works for couple of seconds and then breaks. 

One solution would be to echo the /pose topic when localization starts but no luck. It publishes data to /pose topic after localization breaks (after couple of seconds), so data is incorrect.

Thank you again...

0RBalaji, 2025-01-28:
I see you are using:
`ros2 launch slam_toolbox localization_launch.py params_file:=/path/to/mapper_params_localization.yaml`
It needs nav2_params.yaml configuration file, not the mapper_params file.
Check your launch file and see if the node executables match with the config node names

Agri-Mechatronic, 2025-01-28:
Thanks for the reply. This is my localization_launch.py file:

```
import os

from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.substitutions import LaunchConfiguration
from launch_ros.actions import Node
from ament_index_python.packages import get_package_share_directory

def generate_launch_description():
    use_sim_time = LaunchConfiguration('use_sim_time')
    slam_params_file = LaunchConfiguration('slam_params_file')

    declare_use_sim_time_argument = DeclareLaunchArgument(
        'use_sim_time',
        default_value='true',
        description='Use simulation/Gazebo clock')
    declare_slam_params_file_cmd = DeclareLaunchArgument(
        'slam_params_file',
        default_value=os.path.join(get_package_share_directory("slam_toolbox"),
                                   'config', 'mapper_params_localization.yaml'),
        description='Full path to the ROS2 parameters file to use for the slam_toolbox node')

    start_localization_slam_toolbox_node = Node(
        parameters=[
          slam_params_file,
          {'use_sim_time': use_sim_time}
        ],
        package='slam_toolbox',
        executable='localization_slam_toolbox_node',
        name='slam_toolbox',
        output='screen')

    ld = LaunchDescription()

    ld.add_action(declare_use_sim_time_argument)
    ld.add_action(declare_slam_params_file_cmd)
    ld.add_action(start_localization_slam_toolbox_node)

    return ld
```

This is my nav2_params.yaml file that I use for navigation:

```
amcl:
  ros__parameters:
    use_sim_time: True
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
[... 303 lines cut ...]
  ros__parameters:
    use_sim_time: True
    loop_rate: 20
    stop_on_failure: false
    waypoint_task_executor_plugin: "wait_at_waypoint"
    wait_at_waypoint:
      plugin: "nav2_waypoint_follower::WaitAtWaypoint"
      enabled: True
      waypoint_pause_duration: 200

velocity_smoother:
  ros__parameters:
    use_sim_time: True
    smoothing_frequency: 20.0
    scale_velocities: False
    feedback: "OPEN_LOOP"
    max_velocity: [0.26, 0.0, 1.0]
    min_velocity: [-0.26, 0.0, -1.0]
    max_accel: [2.5, 0.0, 3.2]
    max_decel: [-2.5, 0.0, -3.2]
    odom_topic: "odom"
    odom_duration: 0.1
    deadband_velocity: [0.0, 0.0, 0.0]
    velocity_timeout: 1.0
```

dan9thsense, 2025-04-15:
Maybe it doesn't matter but in your mapper_params_localization.yaml,  you have `mode: "localization"` while the example files do not have quotes. Actually, the mapper_params_localization.yaml file in this repo uses `mode: mapper` which I think is a bug, expected `mode: localization`

olaghattas, 2025-06-26:
The parameter name you're using is incorrect. it's not params_file, it's slam_params_file.

the correct command is  ``` ros2 launch slam_toolbox localization_launch.py slam_params_file:=/path/to/mapper_params_localization.yaml ``` not  ```
ros2 launch slam_toolbox localization_launch.py params_file:=/path/to/mapper_params_localization.yaml```

Gior26, 2025-07-08:
Might be unrelated but make sure to double check that there is no amcl node running as it can interfere with the localization_node from slam_toolbox.