SteveMacenski/slam_toolbox #745: Make the slam_toolbox async node pausable

fbottarel, 2025-01-16:
#### Feature description
As stated in the README, `async_slam_toolbox_node` is not "pausable", i.e. there is no way to prevent the node from processing new LaserScan messages. On our robot, however, we need the navigation and mapping stack to start in a paused state and activated later. We are currently using Humble so the node is not a LifecycleNode. 

#### Implementation considerations
The most appropriate (and hard) way would be to backport the LifecycleNode-ess of the slam_toolbox nodes into the Humble branch.
A simpler way I found looking at the code is changing  

https://github.com/SteveMacenski/slam_toolbox/blob/a69c0fea1d66ff65893e022df950adb58f2a165e/src/slam_toolbox_async.cpp#L56

into

```c++
  // if not paused, process the scan
  if (shouldProcessScan(scan, pose)) {
    addScan(laser, scan, pose);
  }
```

Recompiling `slam_toolbox_node_async` with this change seems to obtain the desired feature.

SteveMacenski, 2025-01-16:
Seems reasonable! 

Please open a PR against humble since you need it, jazzy, and ros2/rolling so its reflected in all new distributions 🙂

fbottarel, 2025-01-17:
Great! I'll open a first PR for `humble`, if that gets approved I'll open one for `ros2` and `jazzy` as well with the same changes (provided it still works).

closed 2025-01-21 by SteveMacenski