SteveMacenski/slam_toolbox #750: slam_toolbox node with a namespace ignores parameter file

manuelarling, 2025-02-07:
**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - binaries
- ROS Version
  - ROS2 Humble
- Version or commit hash:
  - branch humble (4a160dda4d253ba990432810d5e06d40a03d7639)
- Laser unit:
  - Ouster OS2 32
  
#### Steps to reproduce issue
- Start a gazebo simulation with a namespaced roboter and prefixed frame_ids. Start a SLAM node (`async_slam_toolbox_node`) and add the following remappings to the slam node (node started inside the `slam_toolbox.launch.py`):
```python
remappings=[''
            ('/tf', '/robot1/tf'), 
            ('/tf_static', '/robot1/tf_static'),
            ('/map', '/robot1/map'),
            ('/map_metadata', '/robot1/map_metadata')
            ]
```

`ros topic echo /robot1/map` will return messages with a working map.

**The rqt_graph looks like this:**

[image]

**The tf_tree looks like this:**
[image]

If you now add a `PushRosNamespace(namespace='robot1')` to the `bringup_slam.launch.py` that launches the `slam_toolbox.launch.py`, the slam_toolbox will use the new namespace.
```python

    bringup_cmd_group = GroupAction([
        PushRosNamespace(
            condition=IfCondition(use_namespace),
            namespace='robot1'),

        # Set up SLAM Toolbox launch command
        IncludeLaunchDescription(
            PythonLaunchDescriptionSource(os.path.join(scout_mini_gazebo_launch_dir, 'navigation/slam_toolbox.launch.py')),
            launch_arguments={
                              'use_sim_time': use_sim_time,
                              'autostart': 'True'}.items())
    ])
```

#### Expected behavior
That the parameters from the parameter file are applied. which would result with the following changes:
- The rqt_graph would not have a /scan` topic, instead the`/robot1/slam_toolbox` node would subscribe to the /robot1/scan` topic.
- The new tf_tree would include the `/robot1/map frame
- The output of the terminal would be the same
- The same CeresSolver would be used 

#### Actual behavior

**The new rqt_graph looks like this:**

[image]

**The new tf_tree is no longer ok:**

[image]

With `ros2 topic echo /robot1/map`, **no** message is published.

The output from the slam node in the terminal is also different, I added an extra output log for the `preconditioner_type` of the `solver` and it is different:
**Before (Without PushRosNamespace):**
```shell
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [async_slam_toolbox_node-1]: process started with pid [36947]
[async_slam_toolbox_node-1] [INFO] [1738957120.377107634] [slam_toolbox]: Node using stack size 40000000
[async_slam_toolbox_node-1] [INFO] [1738957120.397022167] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[async_slam_toolbox_node-1] [INFO] [1738957120.397218835] [slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[async_slam_toolbox_node-1] [INFO] [1738957120.397225435] [slam_toolbox]: SCHUR_JACOBI
[async_slam_toolbox_node-1] [WARN] [1738957120.485510752] [slam_toolbox]: minimum laser range setting (0.0 m) exceeds the capabilities of the used Lidar (0.3 m)
[async_slam_toolbox_node-1] Registering sensor: [Custom Described Lidar]
```

**After (With PushRosNamespace):**
```shell
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [async_slam_toolbox_node-1]: process started with pid [36645]
[async_slam_toolbox_node-1] [INFO] [1738957060.851233402] [robot1.slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[async_slam_toolbox_node-1] [INFO] [1738957060.851447309] [robot1.slam_toolbox]: JACOBI
```

To me it looks like my slam params file is no longer used by slam_toolbox when applying a namespace.
This is because it is not using the CeresSolver `SCHUR_JACOBI` specified in the params file, but the default `JACOBI`. And also the `/scan` topic is not changed to the specified `/robot1/scan` topic from the params file.

#### Additional information

**slam params file:**
```yaml
slam_toolbox:
  ros__parameters:

    # Plugin params
    solver_plugin: solver_plugins::CeresSolver
    ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
    ceres_preconditioner: SCHUR_JACOBI
    ceres_trust_strategy: LEVENBERG_MARQUARDT
    ceres_dogleg_type: TRADITIONAL_DOGLEG
    ceres_loss_function: None

    # ROS Parameters
    odom_frame: 'robot1/odom'
    map_frame: 'robot1/map'
    base_frame: 'robot1/base_footprint'
    scan_topic: '/robot1/scan'
    use_map_saver: true
    mode: mapping #localization

    # if you'd like to immediately start continuing a map at a given pose
    # or at the dock, but they are mutually exclusive, if pose is given
    # will use pose
    #map_file_name: test_steve
    # map_start_pose: [0.0, 0.0, 0.0]
    #map_start_at_dock: true

    debug_logging: false
    throttle_scans: 1
    transform_publish_period: 0.02 #if 0 never publishes odometry
    map_update_interval: 5.0
    resolution: 0.05
    min_laser_range: 0.0 #for rastering images
    max_laser_range: 20.0 #for rastering images
    minimum_time_interval: 0.5
    transform_timeout: 0.2
    tf_buffer_duration: 30.
    stack_size_to_use: 40000000 #// program needs a larger stack size to serialize large maps
    enable_interactive_mode: true

    # General Parameters
    use_scan_matching: true
    use_scan_barycenter: true
    minimum_travel_distance: 0.5
    minimum_travel_heading: 0.5
    scan_buffer_size: 10
    scan_buffer_maximum_scan_distance: 10.0
    link_match_minimum_response_fine: 0.1  
    link_scan_maximum_distance: 1.5
    loop_search_maximum_distance: 3.0
    do_loop_closing: true 
    loop_match_minimum_chain_size: 10           
    loop_match_maximum_variance_coarse: 3.0  
    loop_match_minimum_response_coarse: 0.35    
    loop_match_minimum_response_fine: 0.45

    # Correlation Parameters - Correlation Parameters
    correlation_search_space_dimension: 0.5
    correlation_search_space_resolution: 0.01
    correlation_search_space_smear_deviation: 0.1 

    # Correlation Parameters - Loop Closure Parameters
    loop_search_space_dimension: 8.0
    loop_search_space_resolution: 0.05
    loop_search_space_smear_deviation: 0.03

    # Scan Matcher Parameters
    distance_variance_penalty: 0.5      
    angle_variance_penalty: 1.0    

    fine_search_angle_offset: 0.00349     
    coarse_search_angle_offset: 0.349   
    coarse_angle_resolution: 0.0349        
    minimum_angle_penalty: 0.9
    minimum_distance_penalty: 0.5
    use_response_expansion: true

```

**slam_toolbox.launch.py:**
```python

import os

# Import necessary modules from ROS and Launch libraries
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.substitutions import LaunchConfiguration
from launch_ros.actions import Node
from ament_index_python.packages import get_package_share_directory
from nav2_common.launch import RewrittenYaml
from launch_ros.descriptions import ParameterFile

def generate_launch_description():
    # Set up configuration for using simulated time and the SLAM parameters file
    use_sim_time = LaunchConfiguration('use_sim_time')
    slam_params_file = LaunchConfiguration('slam_params_file')

    # Declare the launch argument for using simulation time
    declare_use_sim_time_argument = DeclareLaunchArgument(
        'use_sim_time',
        default_value='true',                 # Default to using Gazebo/simulated time
        description='Use simulation/Gazebo clock'
    )

    # Declare the launch argument for specifying the SLAM parameters file
    declare_slam_params_file_cmd = DeclareLaunchArgument(
        'slam_params_file',
        default_value=os.path.join(get_package_share_directory("scout_mini_gazebo"), 'config', 'slam_toolbox_online_async.yaml'),
        description='Full path to the ROS2 parameters file for the slam_toolbox node'
    )

    # Start the SLAM Toolbox node in asynchronous mode, with specified parameters
    start_async_slam_toolbox_node = Node(
        package='slam_toolbox',
        executable='async_slam_toolbox_node', # Launch SLAM Toolbox in async mode
        name='slam_toolbox',                  
        output='screen',                      # Output log to screen
        parameters=[
          slam_params_file,                  # Load SLAM configuration parameters
          {'use_sim_time': use_sim_time}     # Set simulation time if enabled
        ],
        remappings=[
            ('/tf', '/robot1/tf'), 
            ('/tf_static', '/robot1/tf_static'),
            # ('/scan', '/robot1/scan'),
            ('/map', '/robot1/map'),
            ('/map_metadata', '/robot1/map_metadata')
            ]
    )

    # Create a LaunchDescription instance to hold all actions
    ld = LaunchDescription()

    # Add the declared arguments and node to the launch description
    ld.add_action(declare_use_sim_time_argument)
    ld.add_action(declare_slam_params_file_cmd)
    ld.add_action(start_async_slam_toolbox_node)

    return ld

```

SteveMacenski, 2025-02-07:
Your parameter file needs to have the namespace applied to it. Your parameter file is currently under the node name of `slam_toolbox`, which is no longer your node's name. 

This is not a specific issue for SLAM Toolbox, this is true of all configuration files in ROS - the name of the node must match in the param files to the node being launched. In this case, now `robot1/slam_toolbox`

manuelarling, 2025-02-07:
Ohh man sorry I completely forgot about that, how stupid of me. Thanks that fixed it right away. 👍

closed 2025-02-07 by manuelarling