SteveMacenski/slam_toolbox #753: Question about using slam_toolbox in "3D slam"

wladimirpetrov, 2025-02-13:
Hello.

I understand that this framework is perfect for 2D SLAM, but I just can't help to ask your opinion based on the expertise you have.
I have a question about using the slam_toolbox (that does 2D SLAM) in some sort of pseudo "3D SLAM". 

I have several 2D maps merged into one 3D Occupancy Grid, built by crazyflie.
Now when the crazyflie changes its height I run different launch files that open the existing corresponding map with SLAM ("mode: localization"). But when the height changes I need to open a new map with 2D SLAM. What happens in this case is that these 2 maps (previously opened and currently opened) start overwriting each other (keep blinking in rviz2, 1st map 2nd map etc.).

Obviously this framework has been created for 2D SLAM and it does its work, but still do you think there is a way to "forget" about the previously opened map and start using only the newly opened map?

I tried closing the slam_toolbox process when the drone starts changing the height and then run it again while opening new map, the issue still remains the same. 

Thank you for your time!

SteveMacenski, 2025-02-13:
It should only be processing one map at a time, there's no support for multiple maps, so I'm not sure why the first map would still be visualized (but may not actually be being used in the process?). Are you seeing both of the maps updating? Is it that you're launching a second node is that the first node is still running so they are conflicting? 

Keep in mind a pure localization mode would use the localization node, not just the mode.

wladimirpetrov, 2025-02-20:
Yes you're right, as you said one map is processed at a time. I had one node that kept map data inside by accident. So now when i close the slam node and reopen it again, the new map opens the previous one no longer affects the rviz2.

Thank you very much!

closed 2025-02-20 by wladimirpetrov