SteveMacenski/slam_toolbox #763: LifeCycle and Longlife implementation in ros2 humble

kaleb236, 2025-03-18:
#### Feature description

Hi @SteveMacenski, I'm working on frequently changing environment and I think that SlamToolBox is the best option available. Currently I'm using online_sync + .pose .graph map and it works fine however I want to reset map for the new map creation without restarting all the nodes manually. as mentioned in #561 , lifecycle can solve my issue but it's not available in humble version. I tried to build from source ros2 branch but I was facing a lot of build errors. If you have time check on this?

#### Implementation considerations

SteveMacenski, 2025-03-18:
That is a big ABI breaking change that cannot be backported to Humble due to the risk of breaking current production users that have customizations dependent on the current API and behavior (i.e. if you move to a lifecycle node, then anyone using slam toolbox would break immediately since the node wouldn't process data on startup as they expect).

Your best bet is to fix those build errors or migrate to a more updated ROS 2 distribution, which comes with a number of benefits in SLAM Toolbox, Nav2, and beyond

kaleb236, 2025-03-19:
Thank you @SteveMacenski, I'll migrate to updated ROS2 distribution in the future.

closed 2025-03-19 by kaleb236