SteveMacenski/slam_toolbox #765: Slam tool box crash with   std::bad_alloc error when using a large map

asman100, 2025-03-27:
**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  -  binaries
- ROS Version
  - Ros2 Galactic
- Version or commit hash:
  - 2.5.1-1focal.20221208.074139
- Laser unit:
  - RP Lidar S2
  
#### Steps to reproduce issue
Run Slam toolbox with sync mode using a map size > 100mb

[INFO] [launch]: All log files can be found below /home/pi16/.ros/log/2025-03-25-02-50-51-630201-MOTO-26028
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [async_slam_toolbox_node-1]: process started with pid [26057]
[async_slam_toolbox_node-1] [INFO] [1742867452.406169616] [slam_toolbox]: Node using stack size 40000000
[async_slam_toolbox_node-1] [INFO] [1742867452.478598096] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[async_slam_toolbox_node-1] [INFO] [1742867452.479330544] [slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[async_slam_toolbox_node-1] terminate called after throwing an instance of 'std::bad_alloc'
[async_slam_toolbox_node-1]   what():  std::bad_alloc
[ERROR] [async_slam_toolbox_node-1]: process has died [pid 26057, exit code -6, cmd '/opt/ros/galactic/lib/slam_toolbox/async_slam_toolbox_node --ros-args -r __node:=slam_toolbox --params-file /tmp/launch_params_xamgnl61 --params-file /home/pi16/ros2_ws/install/amrj16_slam_toolbox/share/amrj16_slam_toolbox/config/amrj16_slam_config.yaml --params-file /tmp/launch_params_vzj17zje'].

#### Expected behavior
Localize the robot 
#### Actual behavior
Crash with a  std::bad_alloc error on jetson xavier nx 8gb. however same setup works on other ubuntu pcs
#### Additional information
this is our slam toolbox param file and we are specifying the map file and sime time params from the launch file
Tried changing the stack size to use param but no effect
slam_toolbox:
  ros__parameters:

    # Plugin params
    solver_plugin: solver_plugins::CeresSolver
    ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
    ceres_preconditioner: SCHUR_JACOBI
    ceres_trust_strategy: LEVENBERG_MARQUARDT
    ceres_dogleg_type: TRADITIONAL_DOGLEG
    ceres_loss_function: None

    # ROS Parameters
    odom_frame: odom
    map_frame: map
    base_frame: base_footprint
    scan_topic: /scan_filtered
    use_map_saver: true
    #mode: mapping #localization

    # if you'd like to immediately start continuing a map at a given pose
    # or at the dock, but they are mutually exclusive, if pose is given
    # will use pose
    #map_file_name: test_steve
    map_start_pose: [0.0, 0.0, 0.0]
    #map_start_at_dock: true
    debug_logging: false
    throttle_scans: 1
    transform_publish_period: 0.02 #if 0 never publishes odometry
    map_update_interval: 5.0
    resolution: 0.1
    min_laser_range: 0.0 #for rastering images
    max_laser_range: 30.0 #for rastering images
    minimum_time_interval: 0.5
    transform_timeout: 0.2
    tf_buffer_duration: 30.0
    stack_size_to_use: 40000000 #// program needs a larger stack size to serialize large maps
    #enable_interactive_mode: true

    # General Parameters
    use_scan_matching: true
    use_scan_barycenter: true
    minimum_travel_distance: 0.5
    minimum_travel_heading: 0.5
    scan_queue_size: 1
    scan_buffer_size: 10
    scan_buffer_maximum_scan_distance: 10.0
    link_match_minimum_response_fine: 0.1  
    link_scan_maximum_distance: 1.5
    loop_search_maximum_distance: 3.0
    do_loop_closing: true 
    loop_match_minimum_chain_size: 10           
    loop_match_maximum_variance_coarse: 3.0  
    loop_match_minimum_response_coarse: 0.35    
    loop_match_minimum_response_fine: 0.45

    # Correlation Parameters - Correlation Parameters
    correlation_search_space_dimension: 0.5
    correlation_search_space_resolution: 0.01
    correlation_search_space_smear_deviation: 0.1 

    # Correlation Parameters - Loop Closure Parameters
    loop_search_space_dimension: 8.0
    loop_search_space_resolution: 0.05
    loop_search_space_smear_deviation: 0.03

    # Scan Matcher Parameters
    distance_variance_penalty: 0.5      
    angle_variance_penalty: 1.0    

    fine_search_angle_offset: 0.00349     
    coarse_search_angle_offset: 0.349   
    coarse_angle_resolution: 0.0349        
    minimum_angle_penalty: 0.9
    minimum_distance_penalty: 0.5
    use_response_expansion: true

SteveMacenski, 2025-03-27:
Can you get a trace so we can see the issue? https://docs.nav2.org/tutorials/docs/get_backtrace.html

I'm going to make 2 educated guesses before we have data:
- You're actually running out of memory on the Jetson considering other loads and/or AI GPU using the 8gb - so this is a legit error of constraints not possible
- The process runs out of memory within its bounds to use, which `stack_size_to_use` increases. Your param is set to `40000000` which is ~40MB. If your map is 100MB, that's a problem

asman100, 2025-03-27:
Okay ill get a back trace. My colleague had a look at the memory usage on the jetson and we had 3 gb of memory available. Ive tried increasing the stack_size_to_use param to the equivelant of 150 mb but the node still launches and declares 40000000. The map is so large because we are mapping a company warehouse.

asman100, 2025-03-28:
[gdb (1).txt](https://github.com/user-attachments/files/19509480/gdb.1.txt)

asman100, 2025-03-28:
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x0000fffff7432aac in __GI_abort () at abort.c:79
#2  0x0000fffff764a8bc in __gnu_cxx::__verbose_terminate_handler() ()
   from /lib/aarch64-linux-gnu/libstdc++.so.6
#3  0x0000fffff764820c in ?? () from /lib/aarch64-linux-gnu/libstdc++.so.6
#4  0x0000fffff7648270 in std::terminate() ()
   from /lib/aarch64-linux-gnu/libstdc++.so.6
#5  0x0000fffff7648564 in __cxa_throw ()
   from /lib/aarch64-linux-gnu/libstdc++.so.6
#6  0x0000fffff7648b98 in operator new(unsigned long) ()
   from /lib/aarch64-linux-gnu/libstdc++.so.6
#7  0x0000fffff76e0e44 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_mutate(unsigned long, unsigned long, char const*, unsigned long) () from /lib/aarch64-linux-gnu/libstdc++.so.6
#8  0x0000fffff76e1a24 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_replace_aux(unsigned long, unsigned long, unsigned long, char) () from /lib/aarch64-linux-gnu/libstdc++.so.6
#9  0x0000fffff77a8e64 in boost::archive::basic_binary_iprimitive<boost::archive::binary_iarchive, char, std::char_traits<char> >::load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) ()
   from /lib/aarch64-linux-gnu/libboost_serialization.so.1.71.0
#10 0x0000fffff77a9068 in boost::archive::basic_binary_iarchive<boost::archive::binary_iarchive>::init() ()
 from /lib/aarch64-linux-gnu/libboost_serialization.sError detected on fd 0
error detected on stdin
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x0000fffff7432aac in __GI_abort () at abort.c:79
#2  0x0000fffff764a8bc in __gnu_cxx::__verbose_terminate_handler() ()
   from /lib/aarch64-linux-gnu/libstdc++.so.6
#3  0x0000fffff764820c in ?? () from /lib/aarch64-linux-gnu/libstdc++.so.6
#4  0x0000fffff7648270 in std::terminate() ()
   from /lib/aarch64-linux-gnu/libstdc++.so.6
#5  0x0000fffff7648564 in __cxa_throw ()
   from /lib/aarch64-linux-gnu/libstdc++.so.6
#6  0x0000fffff7648b98 in operator new(unsigned long) ()
   from /lib/aarch64-linux-gnu/libstdc++.so.6
#7  0x0000fffff76e0e44 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_mutate(unsigned long, unsigned long, char const*, unsigned long) () from /lib/aarch64-linux-gnu/libstdc++.so.6
#8  0x0000fffff76e1a24 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_replace_aux(unsigned long, unsigned long, unsigned long, char) () from /lib/aarch64-linux-gnu/libstdc++.so.6
#9  0x0000fffff77a8e64 in boost::archive::basic_binary_iprimitive<boost::archive::binary_iarchive, char, std::char_traits<char> >::load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) ()
   from /lib/aarch64-linux-gnu/libboost_serialization.so.1.71.0
#10 0x0000fffff77a9068 in boost::archive::basic_binary_iarchive<boost::archive::binary_iarchive>::init() ()

asman100, 2025-03-28:
That was the result of our backtrace on the jetson

SteveMacenski, 2025-03-28:
>  from /lib/aarch64-linux-gnu/libboost_serialization.sError detected on fd 0

Is this happening when you serialize the map? I thought this implied that this was happening during live SLAM use. If this is happening, its 100% related to the `stack_size_to_use` since that is why that was added. You should increase this value to be proportional to the map size you need. Boost serialization requires that the graph is put on the stack to dump into the file, so the program needs enough bandwidth in memory access to do so

asman100, 2025-03-28:
This is when launching a sync mode with the map included in the config file using localization mode. I tried increasing up to the stack size to the equivalent of 500 mb to no avail. I will send the launch file with an updated config.

SteveMacenski, 2025-03-28:
That is deserializing the serialized file, the same logic applies. If you set it to increased, do you see that this value is used? I want to say that there may have been a bug in how that parameter was handled that we fixed many years ago. I see that you're on Galactic, so we may not have fixed it on that branch since Galactic has been EOL for years. 

Have you tried a more recent branch?

asman100, 2025-03-28:
on the jetson no unfortunately since it doesn't support newer distros of ubuntu but on other machines it works perfectly whether in humble or galactic. Atleast in the terminal it does quote the new value being used.

asman100, 2025-03-28:
This is the current launch configuration and config file.
Node(
               package='slam_toolbox',
               executable='sync_slam_toolbox_node',  
               name='slam_toolbox',
               output='screen',
               parameters=[{'use_sim_time': False}, slam_toolbox_params,
                       {
               'map_file_name': serialized_map,
               'mode': 'localization'
            }],
        ),

slam_toolbox:
  ros__parameters:

    # Plugin params
    solver_plugin: solver_plugins::CeresSolver
    ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
    ceres_preconditioner: SCHUR_JACOBI
    ceres_trust_strategy: LEVENBERG_MARQUARDT
    ceres_dogleg_type: TRADITIONAL_DOGLEG
    ceres_loss_function: None

    # ROS Parameters
    odom_frame: odom
    map_frame: map
    base_frame: base_footprint
    scan_topic: /scan_filtered
    use_map_saver: true
    #mode: mapping #localization

    # if you'd like to immediately start continuing a map at a given pose
    # or at the dock, but they are mutually exclusive, if pose is given
    # will use pose
    #map_file_name: test_steve
    map_start_pose: [0.0, 0.0, 0.0]
    #map_start_at_dock: true
    debug_logging: false
    throttle_scans: 1
    transform_publish_period: 0.02 #if 0 never publishes odometry
    map_update_interval: 5.0
    resolution: 0.1
    min_laser_range: 0.0 #for rastering images
    max_laser_range: 30.0 #for rastering images
    minimum_time_interval: 0.5
    transform_timeout: 0.2
    tf_buffer_duration: 30.0
    stack_size_to_use: 500000000 #// program needs a larger stack size to serialize large maps
    #enable_interactive_mode: true

    # General Parameters
    use_scan_matching: true
    use_scan_barycenter: true
    minimum_travel_distance: 0.5
    minimum_travel_heading: 0.5
    scan_queue_size: 1
    scan_buffer_size: 10
    scan_buffer_maximum_scan_distance: 10.0
    link_match_minimum_response_fine: 0.1  
    link_scan_maximum_distance: 1.5
    loop_search_maximum_distance: 3.0
    do_loop_closing: true 
    loop_match_minimum_chain_size: 10           
    loop_match_maximum_variance_coarse: 3.0  
    loop_match_minimum_response_coarse: 0.35    
    loop_match_minimum_response_fine: 0.45

    # Correlation Parameters - Correlation Parameters
    correlation_search_space_dimension: 0.5
    correlation_search_space_resolution: 0.01
    correlation_search_space_smear_deviation: 0.1 

    # Correlation Parameters - Loop Closure Parameters
    loop_search_space_dimension: 8.0
    loop_search_space_resolution: 0.05
    loop_search_space_smear_deviation: 0.03

    # Scan Matcher Parameters
    distance_variance_penalty: 0.5      
    angle_variance_penalty: 1.0    

    fine_search_angle_offset: 0.00349     
    coarse_search_angle_offset: 0.349   
    coarse_angle_resolution: 0.0349        
    minimum_angle_penalty: 0.9
    minimum_distance_penalty: 0.5
    use_response_expansion: true

SteveMacenski, 2025-03-28:
Try globally increasing the stack size for programs with `ulimit -s unlimited` and seeing if that helps. If so, then its a stack limitation issue that either the parameter isn't being properly handled or the Jetson needs some other way of increasing that limit. It looks like the Jetson's default is pretty small compared to x86 machines (16x smaller), so it wouldn't surprise me if that caused you an issue

asman100, 2025-03-28:
we tried that still the same issue it seems like the parameter for stack size is not being handled properly atleast on the jetson. I have it set at 500000000 and its reading 40000000.

[INFO] [launch]: All log files can be found below /home/pi16/.ros/log/2025-03-28-18-57-49-505557-MOTO-25522
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [map_server-1]: process started with pid [25527]
[INFO] [sync_slam_toolbox_node-2]: process started with pid [25529]
[INFO] [lifecycle_manager-3]: process started with pid [25539]
[map_server-1] [INFO] [1743184671.908242816] [map_server]: 
[map_server-1] 	map_server lifecycle node launched. 
[map_server-1] 	Waiting on external lifecycle transitions to activate
[map_server-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[map_server-1] [INFO] [1743184671.909851424] [map_server]: Creating
[lifecycle_manager-3] [INFO] [1743184672.063809664] [lifecycle_manager]: Creating
[lifecycle_manager-3] [INFO] [1743184672.091862272] [lifecycle_manager]: [34m[1mCreating and initializing lifecycle service clients[0m[0m
[lifecycle_manager-3] [INFO] [1743184672.184692640] [lifecycle_manager]: [34m[1mStarting managed nodes bringup...[0m[0m
[lifecycle_manager-3] [INFO] [1743184672.194306720] [lifecycle_manager]: [34m[1mConfiguring map_server[0m[0m
[sync_slam_toolbox_node-2] [INFO] [1743184672.520114240] [slam_toolbox]: Node using stack size 40000000
[map_server-1] [INFO] [1743184675.560640576] [map_server]: Configuring
[map_server-1] [INFO] [map_io]: Loading yaml file: /home/pi16/ros2_ws/install/amrj16_map_server/share/amrj16_map_server/config/utimac_environment_map.yaml
[map_server-1] [DEBUG] [map_io]: resolution: 0.05
[map_server-1] [DEBUG] [map_io]: origin[0]: -32.5
[map_server-1] [DEBUG] [map_io]: origin[1]: -25.3
[map_server-1] [DEBUG] [map_io]: origin[2]: 0
[map_server-1] [DEBUG] [map_io]: free_thresh: 0.25
[map_server-1] [DEBUG] [map_io]: occupied_thresh: 0.65
[map_server-1] [DEBUG] [map_io]: mode: trinary
[map_server-1] [DEBUG] [map_io]: negate: 0
[map_server-1] [INFO] [map_io]: Loading image_file: /home/pi16/ros2_ws/install/amrj16_map_server/share/amrj16_map_server/config/utimac_environment_map.pgm
[map_server-1] [DEBUG] [map_io]: Read map /home/pi16/ros2_ws/install/amrj16_map_server/share/amrj16_map_server/config/utimac_environment_map.pgm: 1421 X 1306 map @ 0.05 m/cell
[lifecycle_manager-3] [INFO] [1743184676.943844128] [lifecycle_manager]: [34m[1mActivating map_server[0m[0m
[map_server-1] [INFO] [1743184676.947127456] [map_server]: Activating
[map_server-1] [INFO] [1743184677.027693824] [map_server]: Creating bond (map_server) to lifecycle manager.
[lifecycle_manager-3] [INFO] [1743184677.155132896] [lifecycle_manager]: Server map_server connected with bond.
[lifecycle_manager-3] [INFO] [1743184677.160631936] [lifecycle_manager]: [34m[1mManaged nodes are active[0m[0m
[lifecycle_manager-3] [INFO] [1743184677.163300320] [lifecycle_manager]: [34m[1mCreating bond timer...[0m[0m
[sync_slam_toolbox_node-2] [INFO] [1743184677.252037120] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[sync_slam_toolbox_node-2] [INFO] [1743184677.258668000] [slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[sync_slam_toolbox_node-2] terminate called after throwing an instance of 'std::bad_alloc'
[sync_slam_toolbox_node-2]   what():  std::bad_alloc
[ERROR] [sync_slam_toolbox_node-2]: process has died [pid 25529, exit code -6, cmd '/opt/ros/galactic/lib/slam_toolbox/sync_slam_toolbox_node --ros-args -r __node:=slam_toolbox --params-file /tmp/launch_params_anbu7zah --params-file /home/pi16/ros2_ws/install/amrj16_slam_toolbox/share/amrj16_slam_toolbox/config/amrj16_slam_config.yaml --params-file /tmp/launch_params__u45xvdn'].

asman100, 2025-03-28:
we got it to read the stack size parameter size correctly and even after doing the` ulimit -s unlimited` there is no change

SteveMacenski, 2025-03-28:
Are you sure at that instance you have enough RAM available?

asman100, 2025-03-28:
Yea, it seems it might be an issue with the cached memory not being reallocated when called on by a node. However, we were able to get it to function after reducing the resolution to 0.25. I will further reassess the situation with the memory next week and update.

closed 2025-05-06 by SteveMacenski