SteveMacenski/slam_toolbox #766: [Humble branch] Make SlamToolbox a lifecycle node

arilow, 2025-04-13:
#### Feature description

I've seen this is already done in the ros2 branch. Can we backport it to the humble brunch?

#### Implementation considerations

Cherry-picking this commit: de13553d9f92e3773da4b2ff6eebd1977b9337f6 into the humble branch + a couple of minor changes did the trick for me. I'm going to create a PR with these changes.

arilow, 2025-04-15:
Thanks a lot for the quick response @SteveMacenski !

You say that changing the slam toolbox nodes to be lifecycle nodes would change their default behavior? I would expect them to be activated by default and just have a new service to change their state. I'm not an expert, though, so I might be wrong here.

Regarding creating a new branch, `humble_lifecycle, ' I think it's okay if you think it would help other developers. If it isn't that helpful, then I'd save you the trouble of having to maintain a new branch. I can work with the changes in my fork.

Again, thanks a lot for your willingness to help here

SteveMacenski, 2025-04-15:
Lifecycle nodes are not activated by default, they have to be externally stimulated. While I recently added an `autostart` feature into the Launch API, that would still break any user's launch files until hard changes are made. That breaks ABI/API stability policies and cannot be merged.

I just created a new `humble_lifecycle` branch. If you target your PR there and ping me, I can review there!

pky404, 2025-04-25:
SLAM Toolbox localization node crashes on humble_lifecycle branch (no error log)

Description:
The slam_toolbox localization node crashes immediately on launch when using the humble_lifecycle branch. No logs or error messages are printed—just a silent crash.

Observation:
Crash occurs specifically at this function call:
https://github.com/SteveMacenski/slam_toolbox/blob/0e26c02993457d2c0619052499d1813896761c00/src/slam_toolbox_common.cpp#L134