SteveMacenski/slam_toolbox #770: Noise Appearing Along Robot's Trajectory in 2D Map Generated from Outdoor rosbag

YuseiShiozawa, 2025-04-24:
**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Binaries
- ROS Version
  - ROS1 Noetic
- Version or commit hash:
  - 1.5.7
- Laser unit:
  - Hokuyo 30LX (30m range)
  
#### Steps to reproduce issue

1. Record a rosbag with the Hokuyo LiDAR while driving the robot outdoors using a controller.
2. Play back the rosbag and generate a 2D map using gmapping or slam_toolbox.
3. Observe the map and notice the noise-like artifacts that trace the robot's path.

#### Expected behavior
- A clean 2D map without noise artifacts appearing along the robot's path.
#### Actual behavior
- Noise-like artifacts appear along the robot's path when creating the 2D map from the rosbag.
#### Additional information
- No noise appears when running the same test indoors (2nd floor, Building 2).
- Reducing the playback speed (0.5x, 0.01x) does not resolve the issue.
- The --clock option during rosbag playback does not change the result.
- Cartographer does not show noise artifacts, but gmapping and slam_toolbox do.

[image]

MikeDegany, 2025-05-05:
Hi,
I’ve run into the same issue when using a 2D LiDAR outdoors — especially when the laser hits grass or bushes.
I thought it was just a limitation of 2D LiDARs in outdoor environments. If you find a way to reduce this noise, I’d love to help out or test it. 
Thanks for sharing!

YuseiShiozawa, 2025-05-06:
Hi,
Thanks for your comment! I recently used a long-range 3D LiDAR (capable of detecting up to 200 meters) and converted the point cloud to a 2D laser scan using pointcloud_to_laserscan with the following settings. Then, I used slam_toolbox to generate a map.

Surprisingly, as shown in the image below, most of the noise seems to have disappeared.
I'm not exactly sure why the noise was reduced, but it may have to do with the LiDAR's characteristics or how the conversion was configured.

[image]

SteveMacenski, 2025-05-06:
Awesome, closing then!

closed 2025-05-06 by SteveMacenski

YuseiShiozawa, 2025-05-07:
Hold on, we still have no idea why the noise is happening.

SteveMacenski, 2025-05-07:
It sounds like its sensor related given that a new sensor doesn't have this problem. The description @MikeDegany gave is my analysis of the issue as well -- I thought I actually wrote something up about that to respond to you but it looks like it got lost in the GitHub ether since I don't see it in the thread now.

MikeDegany, 2025-05-07:
@YuseiShiozawa  BTW, aside from the noise in the map caused by outdoor scanning, I also notice what looks like a skid mark that likely is the path your robot moved. This could be that your LiDAR beams are hitting parts of the robot itself, the body, other sensors, or mounts. You might want to limit the LiDAR's FOV to avoid those faults.

YuseiShiozawa, 2025-05-08:
Thanks for the information. However, as shown in the figure below, the /scan data obtained from the LiDAR doesn't seem to contain any noise.

[image]

SteveMacenski, 2025-05-08:
I’m also not seeing alot of noise or blurring in the final map there either! 🙂

YuseiShiozawa, 2025-05-08:
But I want to know the cause!!!!!!!!!!!!!

MikeDegany, 2025-05-08:
I have a better image to showcase the issue you brought up! 

[image]

YuseiShiozawa, 2025-05-08:
Under what conditions is that image taken?

MikeDegany, 2025-05-08:
Two ClearPath Jackal robots with Hokuyo UST10LX on an uneven grass field.

YuseiShiozawa, 2025-05-08:
The grass isn't relevant — isn't it simply possible that noise occurs on open roads?

MikeDegany, 2025-05-08:
@YuseiShiozawa  It's kind of the same thing. If your roads are bordered by straight walls, there wouldn't be much noise. But in most cases, we have bushes, trees, and other vegetation along the roads, which show up as noise on the map. 

@SteveMacenski  I wish SLAM could filter out those single, isolated points and ignore them in the mapping process.