SteveMacenski/slam_toolbox #778: support for kilted kajju

RamanaBotta, 2025-05-29:
Could not compile this package for in ros2 kilted Kajju
Any plans to update this package for this?

SteveMacenski, 2025-05-29:
I’m open to PRs to help the migration! What was the compilation issue? I see some warnings which we should take care of, but no errors.

RamanaBotta, 2025-05-29:
Yes @SteveMacenski , I was bit confused with these warnings. I was able to compile it with few warnings.

RamanaBotta, 2025-05-29:
Could you give few references, so I can contribute to it.

SteveMacenski, 2025-05-29:
You should fork this repo and make a new branch to work on the solution. Then, look over the warnings to understand where they are coming from & resolve them. You may need to recompile from a clean workspace each time you test (`rm -r install/ log/ build/`) in order to see the warnings on consequential builds. 

Once you have something working, push your branch to your fork and open a PR 😄

closed 2025-07-14 by SteveMacenski