SteveMacenski/slam_toolbox #782: scan data is moving when using slam toolbox for mapping

werzingerma, 2025-06-19:
Hi

I'm working with SLAM Toolbox and mapping for the first time and have run into an issue I haven't been able to solve. I've already asked in various forums and communities, but unfortunately no one has had any ideas or solutions so far.

I'm not sure if it's okay to ask this kind of question here — if not, feel free to close the issue.

Thanks in advance!

I have the following Project and Setup and get a moving lidar scan and overlapping maps when letting my robot drive. Am i missing something or am i doing something wrong?

**Setup**

I’m working with a small differential-drive robot called Puzzlebot (https://github.com/ManchesterRoboticsLtd/puzzlebot_ros/tree/main).

Hardware: Jetson Nano running ROS 2 Humble

Wheels are driven via a motor driver board; the robot starts up publishing /cmd_vel and wheel velocities.

LiDAR: RPLIDAR A1, integrated with the sllidar ROS 2 driver (https://github.com/Slamtec/sllidar_ros2).

Odometry: a custom dead_reckoning.py node publishes /odom (https://github.com/ManchesterRoboticsLtd/puzzlebot_ros/blob/main/puzzlebot_ros/dead_reckoning.py). Odometry, velocity and LiDAR data are all coming through fine.

**Goal**

Use slam_toolbox (online asynchronous mode) to build a live map.

Feed that map to Nav2 so the robot can

navigate to goals,

update the map while driving, and

report if a goal becomes unreachable.

**Transforms**

Before launching slam_toolbox I publish two static transforms:

base_link ➜ laser_frame        (LiDAR pose)
base_link ➜ base_footprint     (planar footprint)
(I could set base_frame=base_footprint in the slam parameters, but the static transform should work at my knowledge, and it does—for now.)

Resulting TF-tree:

map → odom → base_link → { base_footprint , laser_frame }

[image]

**Command Order**
```
ros2 run puzzlebot_ros dead_reckoning

sudo chmod 777 /dev/ttyUSB1 (for the lidar)

ros2 launch sllidar_ros2 sllidar_a1_launch.py \
  serial_port:=/dev/ttyUSB1 \
  frame_id:=laser_frame

ros2 run tf2_ros static_transform_publisher 0 0 0.1 0 0 0 base_link laser_frame

ros2 run tf2_ros static_transform_publisher 0 0 0 0 0 0 base_link base_footprint

ros2 launch slam_toolbox online_async_launch.py

ros2 launch nav2_bringup navigation_launch.py \
  use_sim_time:=false
```
Topics after launching everything:

[image]

**Problems or Observations**

- Initial RViz view – all frames start at the same origin
- After sending a goal - The robot drives toward it and Laser scan points move with the robot instead of staying fixed in the map (sometimes the tf moves and sometimes only the scan data does)
- After driving around the original map stays, a new map layer is drawn on top, rotated/shifted and map TF stays at the start position, /odom stops before the goal.

[image]

[image]

[image]

_Also did tests to see of my volicities are right but seems fine for me:_
- Angular:

[image]

- Linear:

[image]

SteveMacenski, 2025-06-19:
is this simulation? Sometimes simulation maps are "too perfect" and don't have enough unique textures to work well. Try the TB3 sandbox map from Nav2 bringup and see if you still see the issue.

werzingerma, 2025-06-20:
No, this is not a Simulation.
I’m running slam_toolbox on a real mobile Robot in my Room at Home.

SteveMacenski, 2025-06-20:
What happens with your odometry alone? Are you sure that this isn't misconfigured? If you put rviz in odom frame and drive it around with slam, does it look smooth, correct, move in the right forward direction for your TF tree, etc?

werzingerma, 2025-06-22:
So I did now test it without the slam_toolbox:

```
ros2 run puzzlebot_ros dead_reckoning

ros2 run tf2_ros static_transform_publisher 0 0 0.1 0 0 0 base_link laser_frame

ros2 run tf2_ros static_transform_publisher 0 0 0 0 0 0 base_link base_footprint

ros2 launch sllidar_ros2 sllidar_a1_launch.py serial_port:=/dev/ttyUSB1 frame_id:=laser_frame

ros2 run teleop_twist_keyboard teleop_twist_keyboard
```

and rviz2 with fixed-map odom:

[image]

and here it is, I did first turning right, then left, drive forward a bid and backwards (u see something that doesn't look fine?):

https://github.com/user-attachments/assets/0a88ddaa-ecb5-4fd1-bd52-87c7a26ea179

werzingerma, 2025-06-22:
also did one with adding the LidarScan to rviz:

https://github.com/user-attachments/assets/2322401a-27e5-44fe-9f0e-818094829b74

werzingerma, 2025-06-23:
so just to let you know if fixed my error after looking at my odometer a bit better and saw that it was rotated 180 degree.

After doing the same to my transformers everything works great :)

closed 2025-06-23 by werzingerma

SteveMacenski, 2025-06-23:
Great! I'm glad to hear, I was just going to respond here but that's basically what I thought the issue would end up being. I'm glad you found that yourself! :-)