SteveMacenski/slam_toolbox #791: Black pixels behind obstacles in SLAM Toolbox map – is this expected behavior?

MFHuniovi, 2025-07-14:
Hi, and thank you for your work on SLAM Toolbox.

I'm currently working with a robot and experiencing some issues with localization. While debugging and ruling out potential causes, I noticed something unexpected in the occupancy grid generated by SLAM Toolbox.

As I understand it, obstacles detected by the LiDAR are represented as black pixels on the map, while free space is white. However, I'm seeing areas behind obstacles that also contain black pixels, suggesting further occupied space beyond what the LiDAR should be able to see (e.g., through walls).

Additionally, I'm noticing that walls are not just a single pixel thick, but several pixels wide in the map. This raises further questions:

- Is it normal to see walls or obstacles represented with multiple black pixels in width?

- Is it normal to find clusters of black pixels "behind" obstacles, even when the LiDAR should not have line-of-sight?

- Could this behavior be related to issues with the EKF, odometry, IMU, or other sensor fusion errors?

- Are there any recommended debugging steps or visualizations to better understand whether this is a mapping artifact or a deeper integration problem?

[image]

[image]

[image]

Thanks in advance for any guidance you can provide.

SteveMacenski, 2025-07-14:
> Is it normal to see walls or obstacles represented with multiple black pixels in width?

To some degree, yes since the sensor you're using isn't perfect and noise-free. Though, those walls are a bit thicker than usual so some tuning there with respect to your odometry, SLAM Toolbox configurations may be necessary. If you're using a cheap-o hobbyist grade lidar, that may be as good as it gets. But for a nice professional SICK, Hokuyo, etc you should be able to do better.

 > Is it normal to find clusters of black pixels "behind" obstacles, even when the LiDAR should not have line-of-sight?

Those are usually glass or mirrors. It is actually seeing beyond those bounds outside the space and/or having the beam reflect around. That's pretty typical and wouldn't worry too much about it. You can just clean up a map later (pretty commonly done) or just leave it if its not doing any harm

MFHuniovi, 2025-08-01:
Sorry for taking so long to reply.
Thanks I will look at how to configure the lidar better and do some testing, do you have any documentation or recommendations to avoid setting parameters with trial and error? My lidar is from Sick

Thanks and I think I could close this ticket because now I just need to configure it to see if I get better results.