SteveMacenski/slam_toolbox #803: Add pose graph event publisher

darthegg, 2025-09-01:
Hello! We discussed the following with @SteveMacenski on Slack, and here’s a brief summary and the implementation direction. @hyunsehyun will add more details about the implementation to this conversation soon.

#### Feature description

Two proposed features for exposing pose graph updates during SLAM:

1. **Event publishing** whenever the pose graph changes.
    - **New node event** when a graph node (vertex) is added.
    - **Loop closure event** when a new edge is created between two nodes.
    - Events carry only minimal information (e.g., node ID, timestamp, initial pose) to keep them lightweight.
2. **Pose graph info update service** to query the latest state of the graph.
    - External servers can request updated node poses and edges (e.g., after optimization).
    - Complements the event stream: events keep external systems reactive, while the service ensures consistency if the graph structure or poses change later.

The intent is **not** to transmit the full pose graph continuously, but to provide hooks so that external systems can synchronize their own information (e.g., semantic data) with the evolving pose graph.

#### Background

The long-term goal is to leverage **semantic data obtained during SLAM** in downstream navigation:

- Semantic data will be assigned to nodes in the pose graph,
- `Nav2` can then **query** semantic data, and
- Path planning can be extended to take semantic constraints into account.

However, it would be too heavy, and require large code changes for `slam_toolbox` itself to handle semantic data extraction and storage. Instead, we propose modular approach:

- `slam_toolbox` only publishes graph events and provides pose update services.
- A **separate semantic server** will subscribe to these events, extract semantic data, assign the semantic data to nodes, and manage semantic data storage, modification and query.

This server-side functionality will be developed experimentally in the `navigation2_auxiliary` repository (We will add the ticket link here soon).

#### Implementation considerations

- Introduce minimal ROS2 messages (e.g., `GraphNodeEvent`, `LoopClosureEvent`) and services (e.g., `GetGraphInfo`, `GetUpdatedPoses`).
- Keep code changes minimal:
    - Publish events only at existing decision points where nodes or loop closures are added.
    - Implement pose update services with existing pose graph data structures.
    - Avoid modifying core graph or solver logic.
- Event messages should be **lightweight**, including only the graph information necessary for storing semantic data.
    - Avoid adding overhead to the SLAM loop.

darthegg, 2025-09-08:
> The other option is to send the loop closure event as just a signal and we publish a `pose_graph` topic that contains the entire graph.

**This option seems better.** It would be less confusing for users, and it could also be helpful if someone wants to use the `pose_graph` topic for purposes other than ours. In that case, the `pose_graph` topic would take on the role of a “getter” service, and the "getter" service would no longer be necessary. It also seems good not to include the service, since that could reduce potential confusion with `/slam_toolbox/serialize_map` service.

I’d like to get agreement on one point: in this case, I think **we should publish the `pose_graph` topic not only on loop closure events but also on new node events**. It could be hard to recognize that the `pose_graph` topic is updated only on loop closures unless users have read all the documentation very carefully.

The concern is the additional resources required to publish the `pose_graph` topic this frequently. **How about we proceed with this approach and include test results in the PR to evaluate the actual impact?**

SteveMacenski, 2025-09-10:
I think a draft PR would be the next steps with that content implemented from the last comment! 

> NewNodeEvent.msg

Add in the pose :-) 

> bool occurred

Not sure this is necessary. Just by publishing we know it occurred. The affected IDs and so on will be virtually all of them in some way or another, so that seem like TMI and the user should look at the newly published graph. The loop closure event may just be a standard trigger / bool message that it happened only? 

> GraphNode.msg

I don't think this needs a timestamp, I'm not sure what that adds